OpenEarable/open-earable #14: usability improvements

TobiasRoeddiger, 2023-08-30:
- [ ] BLE characteristic für button press: BLE notify if button was pressed, notify again if it was pressed again, etc.
- [ ] section 1: general setup, section 2: BLE API Usage, section 3: low level functions
- [ ] audio player: send wav file name via BLE, if emptry string stop audio, remove integer logic for playing files
- [ ] ignore mic collision
- [ ] constant freq tone, remove if it causes problems
- [ ] clearly document the BLE API, even though it is edge-ml, it should be in our docs too
- [ ] remove the last column in the config table

Xraydylan, 2023-08-30:
I found a cleaver way to leave freq tone still in.

For the simple usage it can be disregarded.
But if any more complex stuff is needed, it can be used.

I will not mention the freq tone in the BLE API Usage section and only mention in together with the low level functions and further functionality.

TobiasRoeddiger, 2023-08-31:
[OpenEarable_open-earable at v4_experimental_mess.pdf](https://github.com/OpenEarable/open-earable/files/12488847/OpenEarable_open-earable.at.v4_experimental_mess.pdf)

TobiasRoeddiger, 2023-08-31:
[OpenEarable_open-earable at v4_experimental_mess_comments.pdf](https://github.com/OpenEarable/open-earable/files/12489022/OpenEarable_open-earable.at.v4_experimental_mess_comments.pdf)

closed 2023-09-05 by TobiasRoeddiger