OpenEarable/open-earable #75: Burning bootloader

knightron0, 2023-11-13:
Are there steps available anywhere to burn [this bootloader](https://github.com/arduino/ArduinoCore-mbed/tree/main/bootloaders/nano33ble) onto an OpenEarable device that we built ourselves? The webpage mentions briefly about flashing the bootloader and _then_ uploading the latest firmware via Arduino, but we were wondering if there's a more detailed README available!

TobiasRoeddiger, 2023-11-14:
Hi, you‘ll need a J-Link Programmer. Then connect the Jlink to the 4 pads on the back of the main pcb and power OpenEarable via USB (make sure it‘s switched on). Then you can flash the bootloader using JFlash-Light from your computer.

closed 2023-11-14 by TobiasRoeddiger

knightron0, 2023-11-14:
Just to confirm, does it matter if we specifically use a J-Link programmer to do this? [This tutorial ](https://docs.arduino.cc/hacking/software/Bootloader) mentions that we can use an [AVR-ISP](http://www.atmel.com/dyn/products/tools_card.asp?tool_id=2726) (in-system programmer), [USBtinyISP](http://www.ladyada.net/make/usbtinyisp/) or build a [ParallelProgrammer](https://docs.arduino.cc/hacking/hardware/ParallelProgrammer). We also have an ST-Link programmer available so were wondering if there's a specific reason that the J-Link programmer is recommended?

TobiasRoeddiger, 2023-11-15:
The ST-Link works as it supports SWD. You can also use something like this: https://www.waveshare.com/arm-debugger.htm

knightron0, 2023-12-27:
Using an ST-Link programmer to program the chip is not well documented (see [this](https://electronics.stackexchange.com/questions/250664/can-i-use-st-link-programmer-for-non-st-chips)). 

Secondly, what is the software to be used with the ARM debugger? I can't seem to get it to work with J-Flash or the Keil IDE.

TobiasRoeddiger, 2023-12-27:
We recommend using J-Link, we do not support the other options, sorry.

knightron0, 2024-01-08:
yep — it worked, thank you! just another question though — how does the microphone interface work? i can't seem to find any recordings on my SD card (it's exFAT formatted and the recommended one). is there a way via arduino/the dashboard i can just record audio?

TobiasRoeddiger, 2024-01-12:
Have you tried using the record audio feature of the dashboard? https://openearable.github.io/dashboard/

knightron0, 2024-01-12:
The recorder section on the bottom right? Yes, I tried turning that on multiple times but all but once it just gave me an empty CSV file. I haven't gotten audio through that though, just some sensor readings (which were all constant for some reason).

TobiasRoeddiger, 2024-01-17:
Hi, Sorry for the confusion:

- The recorder in the bottom right is just to record pressure and temperature data in the browser.
- For the microphone, just set the sampling rate and it will automatically start recording that to the sd card. Make sure to disabled the mic after you have recorded to avoid corrupted files.

knightron0, 2024-01-17:
Thank you — seems like it worked!

knightron0, 2024-01-22:
oops, really sorry for the constant questions but i just had a few more. i was trying to get audio data over BLE through the dashboard. however, when i start recording through the dashboard, i don't seem to collect any data (pressure/accelerometer) via BLE. 

my plan was to just save the PDM mic data in the `subscribeOnSensorDataReceived` in `RecordingManager.js` but it seems like it's not being executed. any tips to debug? is there a better way to collect audio data through BLE or an example i can try to verify that it works?

TobiasRoeddiger, 2024-01-23:
The current version of OpenEarable does not have wireless audio streaming.

knightron0, 2024-01-26:
got it, i'll try and see how far i get. in the meanwhile, could you give some pointers on how to debug the blank recording issue? like the pressure etc sensors also don't return any data?

ooshyun, 2025-02-03:
Hey, I am developping this open-earable device, and a beginner to burn the bootloader.

I found the pads on the backside of the PCB(3V3, GND SWCLK, and SWDIO), but it seems to solder those to the wire considering [PCB design](https://oshwlab.com/openearable/openearable-v1_copy_copy_copy_copy_copy) and connect those to Jlink pins. Can you let me know how to connect 4 pins to j-link?

TobiasRoeddiger, 2025-02-03:
We recommend a PCBite to connect to the pads: https://sensepeek.com/pcbite-kit-with-4x-sq10-probes-and-test-wires

ooshyun, 2025-02-08:
Thanks, It worked well by using PCBite!