OpenEarable/open-earable #80: Unable to detect SD card

Oscar-Chung-CKL, 2024-03-18:
Dear all,

I am writing a simple testing code for the board, which is for detecting if an SD card is inserted and available. However, it seems that the code is not working ever with my SD card already inserted into the slot. I worry that it is becuase of the pin number setting issue. This is the code i am using:
```

#include <BufferedPrint.h>
#include <FreeStack.h>
#include <MinimumSerial.h>
#include <RingBuf.h>
#include <SdFat.h>
#include <SdFatConfig.h>
#include <sdios.h>
#include <SPI.h>

const int SPICLK = 52;    // SPI Clock pin
const int SPICS = 51;     // SPI Chip Select (CS) pin
const int SPIMISO = 50;   // SPI MISO (Master In Slave Out) pin
const int SPIMOSI = 48;   // SPI MOSI (Master Out Slave In) pin
const int SD_STATE = 5;   // SD Card State pin

SdFat sd;

void setup() {
  Serial.begin(9600);
  while (!Serial);
  Serial.println("Started");

  pinMode(SPICLK, OUTPUT);
  pinMode(SPICS, OUTPUT);
  pinMode(SPIMISO, INPUT);
  pinMode(SPIMOSI, OUTPUT);
  pinMode(SD_STATE, INPUT);

  if (digitalRead(SD_STATE) == HIGH) {
    Serial.println("SD card is inserted");
  } else {
    Serial.println("No SD card is inserted");
  }

  if (!sd.begin(SPICS, SPI_FULL_SPEED)) {
    Serial.println("No SD card detected");
    while (1);
  }

  Serial.println("SD card detected");
}

void loop() {
  // No action
}

```
Especially the part of:
```
const int SPICLK = 52;    // SPI Clock pin
const int SPICS = 51;     // SPI Chip Select (CS) pin
const int SPIMISO = 50;   // SPI MISO (Master In Slave Out) pin
const int SPIMOSI = 48;   // SPI MOSI (Master Out Slave In) pin
const int SD_STATE = 5;   // SD Card State pin
```

May i know if the pin setting is correct or not, or there is some other issues ?

Thank you.

TobiasRoeddiger, 2024-03-19:
@mkuettner97 can you have a look at this?

mkuettner97, 2024-03-22:
Hi,

all currently used pins are declared inside "Earable_Pins.h".

// SPI Pins
#define EPIN_SPI_CS     27
#define EPIN_SPI_CLK    26
#define EPIN_SPI_MOSI   8
#define EPIN_SPI_MISO   34

// SD State Pin
#define EPIN_SD_STATE   22

Note that SD_STATE should be declared as INPUT_PULLUP instead of just INPUT.

Best Regards,
Michael

closed 2024-03-25 by TobiasRoeddiger