OpenEarable/open-earable #81: Bluetooth sensor data

knightron0, 2024-03-19:
when i run the openEarable with the dashboard, the CSV that i generate after clicking "stopping recording" is always empty. i was looking into more how the dashboard and the openEarable communicate through BLE, and i couldn't find any code that actually writes any IMU/BME280 data to the Bluetooth. 

for example, the [BLE config](https://github.com/OpenEarable/open-earable/blob/main/src/ble_config/ble_config_earable.cpp) file does not have UUIDs for sensor data as mentioned in the Readme, and there isn't an equivalent file to the [Button Service](https://github.com/OpenEarable/open-earable/blob/main/src/button_service/Button_Service.cpp) for sensors anywhere.

am i missing something? let me know!

TobiasRoeddiger, 2024-03-19:
First set the sensor configuration (the graphs will start plotting data), then start the recording.

The sensor uuid is implemented in the edge-ml library used internally by OpenEarable.

closed 2024-03-19 by TobiasRoeddiger

knightron0, 2024-03-21:
sorry for the constant questions -- but i can't seem to get the sensor data to work through BLE. i tried setting the configuration but the graphs stay empty and the BLE handler for the data inside the JS file doesn't get called. 

do you have any tips on how to debug this? i have a bunch of debug statements in both the OpenEarable and EdgeML arduino libraries, but i'm not able to get any output/data from files like `IMU_Sensor_Nano.cpp` inside `boards/generic_boards/ble33nano` or even `generic/BLEHandler_G.cpp`. is there a way to properly debug this? 

thank you so much for the quick replies!

TobiasRoeddiger, 2024-03-25:
It must be either:

1) Something went wrong while following the guide to compile the firmware. Just in case I have uploaded the precompiled firmware which you can flash via SWD and JFlash Lite: https://github.com/OpenEarable/open-earable/tree/main/resources/precompiled

2) Your hardware is broken. Based on our records I presume you built it yourself? If you want to be sure that your hardware I recommend getting online at https://shop.openearable.com/products/openearable-v1-3-starter-bundle

knightron0, 2024-03-27:
seems like something was wrong with the IMU sensor -- the program crashed everytime inside `IMU_Sensor.cpp` when `IMU->begin()` was called. i can see the graphs on the dashboard for the pressure/temperature sensors, but it's all 0s right now. 

it doesn't matter that much for the IMU/temperature sensors, but i just needed to confirm that the Bluetooth interface for sending data was working. thank you so much!